iiot2k/gpiox #1: node red not reacting when more than three gpiox-in

mhin76, 2024-06-14:
I noticed, that when I use three gpiox-in all works just fine (in addition I use several gpiox.out). 
but as soon as I add another gpiox-in, nodered hangs on the second deployment! Means, first deployment all fine, when I change anything in node red and I want to reploy it is not reacting anymore and I have to kill the node red session, set that flow to disabled via flows.json file and restart node red again.
I do not see any message on node red debugging (on info level).

is there a max on those listener? any idea how I could further debug?

iiot2k, 2024-07-07:
Hello,
sorry for the late reply.
That is really very interesting.
Thank you for your tip.
I'll try to recreate it.
Can you also tell me the version of the OS, node.js and Node-Red?
Best regards
Viele Grüße
Derya

mhin76, 2024-07-08:
I run bookworm (Linux kernel 6.6.31) on a RP5 with NodeRed v3.1.9 and node.js v20.15.0.
Sonnige Grüsse, Martin

FlamingFury321, 2024-07-08:
I have the same issue.
The selected pin doesn't matter but as soon as I enable the 4th GPIO Node (even when not connected) it hangs and i cannot deploy anything anymore. After 30 seconds I get an "Error: connect ETIMEDOUT" error on one of my mysql nodes (even though it isn't connected in any way to the GPIO node).
LG Daniel

iiot2k, 2024-07-08:
Hallo Martin,
Hallo Daniel,
habe gestern meine Kündigung erhalten.
Muss mich etwas gedanklich beruhigen.
Aber natürlich muss dieser Fehler behoben werden.
Wenn Ihr wollt, kann ich euch die Quelldateien zusenden.
Schreibt mir zu iiot2k @ gmail . com
LG Derya

mhin76, 2024-07-09:
yes, exactly same behavior. I have then to disable that flow via nano flows.json and restart node red to get in again. Attached my flow.
[flows (3).json](https://github.com/user-attachments/files/16142757/flows.3.json)

those 5 at the bottom I tried to enable. Three of them I kept disabled.

No hurry about the fix. for me not so urgent at least :)

mhin76, 2024-07-09:
Hallo Derya

 

Oohhhhh, das tut mir leid zu hören!! Ich wünsche dir viel Erfolg bei der nächsten Tür, die dadurch aufgeht!!

Fix ist überhaupt nicht eilig für mich.

 

Sonnige Grüsse, Martin

iiot2k, 2024-07-17:
## Fix watch_gpio 🐛
>The watch_gpio function only accepts a few gpio watches.<br>
The reason is that the gpio watches run in the background as a thread.<br>
Unfortunately, the maximum number of threads is set to 4.<br>
One solution is to increase this number of threads.<br>
With the command:<br>
```
export UV_THREADPOOL_SIZE=1000
```
>You can increase this number of threads (max. 1024) for node.js.<br>
Of course, this number of threads also increases the memory requirement.<br>
Try other numbers of threads too.<br>
This also applies to the Node-Red node.<br>
All other functions work correctly.<br>

iiot2k, 2024-07-18:
Solved with Version 2.1.3
Thanks to all who helped.

closed 2024-07-18 by iiot2k

mhin76, 2025-01-01:
@iiot2k what means solved in version 2.1.3? I updated to 2.1.4 and have enabled more than 5 GIPO input but then node red cannot be reloaded anymore via webpage. I also added the "export UV_THREADPOOL_SIZE=64" in my .bashrc and restarted nodered, but no change. Do I miss something? CPU is low (10%), memory same (25% used). Input seem to work, just nodered is not reloading on UI. Nothing under node-red-log (Info level).
PS: HAPPY NEW YEAR!!

iiot2k, 2025-01-01:
Hi Martin,

Are you using 32bit or 64bit OS?

Derya

iiot2k, 2025-01-01:
Hi Martin,
Hi Daniel,
habe auch vor die Entwicklung von Node-Red Nodes einzustellen und nur an den node.js Bibliotheken weiterzuarbeiten.
Die Gründe dafür sind, da es mit zuviel wird an all der Menge zu arbeiten.
Wie gesagt, werde ich auch nur 64bit OS unterstüzen.
Auch einige User im Node-Red Forum gehen mir echt auf den Keks mit Ihren Angriffen gegen mich.
Denke, dass Ihr dafür Verständnis habt.

Viele Grüße
Derya

mhin76, 2025-01-02:
Hallo Derya
Ich verwende die 64bit Version. Ja, habe ich auch bei anderen Projekten mitbekommen, dass es einfach keine Wertschätzung gibt und die Benutzer nur fordern. Sehr sehr schade :-( 
Wenn ich Zeit habe, lese ich mich mal mehr in dieses Thema ein. Hab schon geschaut, ob es IO Input via USB gibt um diese neue Architektur von RP5 zu umgeben. 

Herzliche Grüße und guten Start ins Neue Jahr

Martin

iiot2k, 2025-01-02:
Hallo Martin,
vielen Dank für eine Antwort und ich wünsche dir auch ein gutes und erfolgreiches 2025.
Wie gesagt, muss ich auf vielen Feldern kämpfen und habe auf Node-Red keine Lust mehr.
Es reicht mir schon Node.js in den Griff zu bekommen, weil dieser einfach nicht thread fähig ist.
Aber manche Sachen müssen im Hintergrund laufen, damit das ganze einigermaßen auch flüssig läuft.
Und dann kommt auch noch die Eigenarten von Node-Red dazu.
Du müsstest mal den C++ Code sehen, welche Klimmzüge ich anstellen muss, um nicht Node.js zu blockieren.
Aber ich komme von der Real-Time Welt und das ist echt eine Herausforderung.

Momentan bin ich an der Entwicklung der Nodes für den Baustein MCP23017.
Das ist ein 16 Kanal I/O Port Erweiterung angesteuert über i2c.
Ist zwar nicht so schnell wie das GPIO vom Rpi, ist aber sehr gesund für den Rpi wenn mal was schief läuft.

Viele Grüße
Derya

Media250, 2025-01-03:
Hi, just wanted to say thanks for the module and the PWM function is exactly what i need.
I am also having the same issues with multiple gpiox nodes in and then having them not respond after a few deploys.
(running updated raspbian on a pi4 with gpiox version 2.1.4)
I understand you're working on the issue but can there be a bug list on the NPM or node-red library page?
I spent a lot of time troubleshooting and ended up stumbling on this thread. Would be great to get updates!

Thanks!