iiot2k/gpiox #2: GPIO doesn't work on Pi Zero 1 or 2

supercrab, 2024-07-03:
I've used the library in a Volumio plugin.  It's been tested on a Pi 5 and Pi 4 and it works fine.  However, on my Pi Zero 2 and Pi Zero, setting GPIO doesn't seem to do anything.   

```
gpiox.set_gpio(OUTPUT_PIN, 0);
```

What could be the issue?

iiot2k, 2024-07-07:
Hello,
that's very interesting.
Unfortunately I don't have a Pi Zero 2 or Pi Zero to test.
But I'll order one to test.
The gpiox library uses the official character interface from Linux.
The Pi Zero 2 and Pi Zero have a different CPU.
It may be that this interface is not implemented for the Pi Zero 2 and Pi Zero.
But I can't tell you that until I've tested it.
Can you also tell me which OS and node.js version you are using the library on?
Derya

supercrab, 2024-07-07:
Yes, I think you're right about the interface.  Here are the details: ☺

```
Volumio Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Welcome to Volumio for Raspberry Pi (5.4.83-v7+ armv7l)
volumio@volumio3:~$ node -v
v14.15.4

volumio@volumio3:~$ hostnamectl
   Static hostname: volumio3
         Icon name: computer
        Machine ID: 29f71c0343f0889aad6c0b1561a903be
           Boot ID: 4f4b087dee2544c6bbaccc80a0ea907a
  Operating System: Raspbian GNU/Linux 10 (buster)
            Kernel: Linux 5.4.83-v7+
      Architecture: arm
```

Darmur, 2024-09-13:
any update on this?

foonerd, 2024-10-18:
@iiot2k - what test can we do to help you troubleshooting?

closed 2024-10-27 by iiot2k

iiot2k, 2024-10-27:
I bought a Raspberry Pi Zero W and tested gpio.
I used the latest 32-bit version (Bookworm) as the OS.
When compiling the example program, I found an error in the Makefile for the Raspberry Pi Zero W.
I will correct this in the next version:

```
UNAME_M := $(shell uname -m)
ifeq ($(UNAME_M),aarch64)
LIBFLAG := -L. -lgpiox_arm64
else
LIBFLAG := -L. -lgpiox_arm32
endif

CFLAGS := -std=c++14 -pthread

all: $(patsubst %.cpp, %, $(wildcard *.cpp))

%: %.cpp Makefile

g++ $< -o $@ $(CFLAGS) $(LIBFLAG)

```
The c++ example programs from gpiox all worked on the Raspberry Pi Zero W.

The gpiox library uses the standard GPIO interface from Linux.

If the GPIO interface is not implemented by Linux, gpiox cannot work properly.

Please understand that I cannot test all combinations of Raspberry Pi and operating system.

Especially like Homeassistant or Volumio...

I have neither the time nor the money for that.

Sorry for that

Darmur, 2024-10-27:
thanks for checking. when we can expect the new version with the fix for Raspberry Pi Zero W?

supercrab, 2024-10-27:
Thanks for your help!  I'll test this on volumio when the new version is released. I'm sure it'll be fine 😊