iiot2k/gpiox #4: "chip error"

gearloose83, 2024-09-17:
I'm using a Raspberry Pi 3B+ and the latest Node-Red. The Node worked perfectly fine, but as i played around with some other Nodes and Settings, it stopped working. So i reinstalled Node-Red and your gpiox-Node, but i can't get it to work and the only error-message i'll get is "chip error" and i don't know what to do... 
It seems to make no difference wich pin i try to use, the error ist the same all the time (at the Moment i try to work with gpio13, wich worked fine before

gearloose83, 2024-09-17:
I also tried it with a Pi4, a fresh install and only a inject-node and the gpiox pwm node. After deploying it says "chip error".
Could it be, that this issue came with the last update?

ShkolaIgor, 2024-09-18:
I have same mistake on PI4 after update

ShkolaIgor, 2024-10-06:
Install latest gpiod:

$ sudo apt-get update
$ sudo apt-get install -y curl
$ curl -SL --output setup-libgpiod.sh https://raw.githubusercontent.com/devdotnetorg/docker-libgpiod/HEAD/setup-libgpiod.sh
$ chmod +x setup-libgpiod.sh
$ sudo ./setup-libgpiod.sh

Select  (4) Installation from binaries .

suprimex, 2024-11-09:
> Install latest gpiod:
> 
> $ sudo apt-get update $ sudo apt-get install -y curl $ curl -SL --output setup-libgpiod.sh https://raw.githubusercontent.com/devdotnetorg/docker-libgpiod/HEAD/setup-libgpiod.sh $ chmod +x setup-libgpiod.sh $ sudo ./setup-libgpiod.sh
> 
> Select (4) Installation from binaries .

I got: 
ERROR. Current OS raspbian not supported.

My system:
Linux rpi3b 6.6.47+rpt-rpi-v7 #1 SMP Raspbian 1:6.6.47-1+rpt1 (2024-09-02) armv7l GNU/Linux

iiot2k, 2024-11-09:
Like libgpiod, gpiox uses the official character device interface uApi from Linux. If this is not implemented in Linux, gpiox and libgiod cannot work. On the official Raspberry Pi OS, gpiox works without errors. Unfortunately, I cannot test gpiox on all Linux variants and Raspberry Pi models. I think you will understand.

suprimex, 2024-11-10:
It is the official Raspberry Pi OS running on Raspberry 3B...

iiot2k, 2024-11-10:
Unfortunately my last RPi 3B went up in smoke. I'll try to get a RPi 3B and test it.

iiot2k, 2024-12-03:
Is fixed and online.
Please update.
Can you please test again with the new version and let me know here?
Thanks

iiot2k, 2024-12-03:
Since kernel 6.6.45, the chip on the Rpi 5 is also addressed with /dev/gpiochip0. Before this version, /dev/gpiochip4 is valid. Thank you Raspberry for this confusion.

closed 2024-12-08 by iiot2k